rzeldent/esp32-smartdisplay #163: how to i get touch event so i can control backlight after a time period?

d4rkdpg, 2024-05-01:
how to i get touch event so i can control backlight after a time period?  i.e if a time period passes turn off back light and if touched turn on.

thanks

olivluca, 2024-05-01:
This is how I do it
https://github.com/olivluca/TrumaDisplay/blob/52364f48db71ebd708842bf2f8ce596cee0b0c3d/src/main.cpp#L203
https://github.com/olivluca/TrumaDisplay/blob/52364f48db71ebd708842bf2f8ce596cee0b0c3d/src/main.cpp#L235

Note that not only I turn off the backlight, but I also switch to an empty screen (so that the first touch event doesn't trigger anything, apart from switching back to the main screen).

d4rkdpg, 2024-05-03:
thanks i will give it a shot

d4rkdpg, 2024-05-03:
works fine perfect

closed 2024-05-03 by d4rkdpg