rzeldent/esp32-smartdisplay #167: 4827S043C GT911 Interrupt Pin used?

Danez-DK, 2024-05-24:
Hi,
This is more of a question, than an issue.

As far as I can understand, most 4827S043C boards does not have the Interrupt pin "enabled" by default.
You would have to bridge R17 (connect GT911 to GPIO18), and might have to remove/replace R5, as the GT911 might be too weak to pull GPIO18 low. Also the XPT2046 (U1) chip can be removed, as it is not used for anything, but it is attached to GPIO18.

On theses boards, the program code would have to check if there are changes from the GT911, and use more CPU on that task.

If I make the above changes, should I then change some settings somewhere to make LVGL use interrupt, or does it check if interrupts work, and if not, it just check the GT911 chip for changes all the time?

I could just make the changes, and see what happens, just wanted to ask someone that seems to have a better understanding of how LVGL and hardware input works. :)

You have done a Great job, I really appreciate all the work you have put into this. :)

rzeldent, 2024-06-14:
If you make modifications, you should update the board definition and assign a pin to the IRQ.
There is support for INT in esp_touch_gt911.c but untested...

Good luck!

rzeldent, 2024-09-08:
Little correction here; there is INT support to add a callback in the gt911.c but this is not used.
LVGL uses polling to obtain the touch coordinates.

Danez-DK, 2024-09-08:
Hi,
I got interrupt working, by doing the hardware changes, and adding the interrupt callback.
It gives about 10-15% less CPU load, compared to polling.
But I had to use a powered usb, which supplied more than 0.5A to get it to work.

mpolitowicz, 2025-03-18:
@Danez-DK Can you tell what hardware changes have you implemented ? (details on how INT and RST is connected to CPU)

I came across your comment as we are trying to make INT work for board designed by ourselves using similar chip GT9271. 
We were able to initialize it properly with pooling, but we are struggling with INT (it seems as we are getting constant flicker on INT pin...).