rzeldent/esp32-smartdisplay #177: display frozen/unresponsive

olivluca, 2024-06-24:
This is more a question than a bug report, anyway, it happened to me twice that the display froze and was completely unresponsive.
It's difficult to debug because I don't have a serial port attached and it happens very seldom (twice in a month).
The first time it happened, I added a  [watchdog ](https://github.com/olivluca/TrumaDisplay/commit/4ec0a7f14ceb183643132f1ca934e0d0c4f61d29) to restart the esp32.
I'm quite sure that the esp32 restarted at least once due to the watchdog, however yesterday the screen froze but the watchdog didn't trigger, so I think that either the main loop was still running (so the frozen display is a bug somewhere in the library) or there is an hardware problem.
The board is an esp32-2432S028R.
I'm not 100% sure, but the problem happens after [this line](https://github.com/olivluca/TrumaDisplay/blob/4ec0a7f14ceb183643132f1ca934e0d0c4f61d29/src/main.cpp#L217) (i.e. after an inactivity timeout the brightness is set to 0 and I show a dummy screen to avoid registering clicks on the now invisible objects, when there is new activity that line switches to the main screen). Both time I saw the problem the screen was off, I tapped on it, it showed the main screen then froze.

rzeldent, 2024-09-08:
I suggest to enable LVGL logging and save the logging... Maybe this will help to pinpoint the issue...

olivluca, 2024-09-08:
Thank you, I have no idea how to do that and it if it's even possible without serial :wink: 
Maybe I should upgrade the library first (when I reported this your repo was using lvgl 8.something).
BTW, it didn't happen since last time.

olivluca, 2025-04-26:
I think the problem was that I was making calls to the library from a different thread. Now that I upgraded to lvgl 9.2.2 the problem manifested continuosly. I moved all the calls to the main thread (loop()) and now it seems it doesn't lock up anymore.
I suppose that lvgl is not thread safe (I should know, I've been programming with delphi vcl and lazarus lcl for more than 20 years).

closed 2026-10-09 by github-actions[bot]