rzeldent/esp32-smartdisplay #180: esp32-2432S024C, compilation of the library fails

hezamu, 2024-06-30:
Hi, first of all thanks for this amazing library, it's just what I was looking for.

Now, I'm trying to create a new project, but my compilation fails. Any ideas? I have added the board as a git submodule and `main.cpp` is basically a helloworld. My `platformio.ini` looks like this:

`[platformio]
default_envs = esp32-2432S024C

[env]
platform = espressif32
framework = arduino

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_filters = esp32_exception_decoder

build_flags =
    -Ofast
    -Wall
    '-D BOARD_NAME="${this.board}"'
    '-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_INFO'
    #'-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG'
    #'-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE'
    '-D LV_CONF_PATH=${platformio.include_dir}/lv_conf.h'
    
lib_deps =
    https://github.com/rzeldent/esp32-smartdisplay

[env:esp32-2432S024C]
board = esp32-2432S024C`

Compile log:

`Processing esp32-2432S024C (board: esp32-2432S024C; platform: espressif32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-2432S024C.html
PLATFORM: Espressif 32 (5.0.0) > esp32-2432S024C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 2.0.9+sha.0815c09
|-- lvgl @ 8.4.0
Building in release mode
Linking .pio\build\esp32-2432S024C\firmware.elf
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(lvgl_panel_ili9341_spi.c.o):(.literal.lvgl_lcd_init+0x38): undefined reference to `esp_lcd_panel_disp_on_off'       
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(lvgl_panel_ili9341_spi.c.o): in function `lvgl_lcd_init':
<homedir>\src\platformio\<project>/.pio/libdeps/esp32-2432S024C/esp32_smartdisplay/src/lvgl_panel_ili9341_spi.c:91: undefined reference 
to `esp_lcd_panel_disp_on_off'
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(esp_touch_cst816s.c.o):(.literal.cst816s_read_data+0x8): undefined reference to `esp_lcd_panel_io_rx_param'
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(esp_touch_cst816s.c.o): in function `cst816s_read_data':
<homedir>\src\platformio\<project>/.pio/libdeps/esp32-2432S024C/esp32_smartdisplay/src/esp_touch_cst816s.c:158: undefined reference to `esp_lcd_panel_io_rx_param'
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(esp_touch_cst816s.c.o): in function `cst816s_read_info':
<homedir>\src\platformio\<project>/.pio/libdeps/esp32-2432S024C/esp32_smartdisplay/src/esp_touch_cst816s.c:121: undefined reference to `esp_lcd_panel_io_rx_param'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-2432S024C\firmware.elf] Error 1
================================================================ [FAILED] Took 7.85 seconds ================================================================
 *  The terminal process "<homedir>\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task in folder <project>: <homedir>\.platformio\penv\Scripts\platformio.exe run 

Processing esp32-2432S024C (board: esp32-2432S024C; platform: espressif32; framework: arduino)
[... 304 lines cut ...]
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32-2432S024C\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32-2432S024C\libFrameworkArduino.a
Linking .pio\build\esp32-2432S024C\firmware.elf
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(lvgl_panel_ili9341_spi.c.o):(.literal.lvgl_lcd_init+0x38): undefined reference to `esp_lcd_panel_disp_on_off'       
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(lvgl_panel_ili9341_spi.c.o): in function `lvgl_lcd_init':
<homedir>\src\platformio\<project>/.pio/libdeps/esp32-2432S024C/esp32_smartdisplay/src/lvgl_panel_ili9341_spi.c:91: undefined reference 
to `esp_lcd_panel_disp_on_off'
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(esp_touch_cst816s.c.o):(.literal.cst816s_read_data+0x8): undefined reference to `esp_lcd_panel_io_rx_param'
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(esp_touch_cst816s.c.o): in function `cst816s_read_data':
<homedir>\src\platformio\<project>/.pio/libdeps/esp32-2432S024C/esp32_smartdisplay/src/esp_touch_cst816s.c:158: undefined reference to `esp_lcd_panel_io_rx_param'
<homedir>/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S024C\libe0d\libesp32_smartdisplay.a(esp_touch_cst816s.c.o): in function `cst816s_read_info':
<homedir>\src\platformio\<project>/.pio/libdeps/esp32-2432S024C/esp32_smartdisplay/src/esp_touch_cst816s.c:121: undefined reference to `esp_lcd_panel_io_rx_param'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-2432S024C\firmware.elf] Error 1
=============================================================== [FAILED] Took 16.92 seconds ===============================================================
`

rzeldent, 2024-07-01:
Looks like functions as: ```esp_lcd_panel_io_rx_param``` are not defined. Did you update to the latest release of the esp libraries?

hezamu, 2024-07-01:
Thank you, that was my problem.

For some reason I assumed that PlatformIO would keep my libraries up to date, but the Espressif 32 platform was kept at version 5.0.0.

closed 2024-07-01 by hezamu

acheronfail, 2025-06-13:
For anyone else having this problem, I found running 

```shell
pio pkg update
```

in my project directory updated everything and fixed the issue for me.