rzeldent/esp32-smartdisplay #186: esp32-8048S050C Wrong colors

xperiments, 2024-07-12:
Cloned the demo repo
Added some gradients to it
5 gradients
Red = BANDING in levels of 64
Green = is OK
Blue = is OK
Yellow  BANDING in levels of 64
Magenta BANDING in levels of 64

Tried anything I know, swaping colors, converting to 565 color nothing...

[image]

Using this code
```
#include <Arduino.h>

#include <esp32_smartdisplay.h>
#include <ui/ui.h>

#include <src/extra/libs/qrcode/lv_qrcode.h>
lv_obj_t *createGradient(lv_obj_t *obj, uint32_t color, uint16_t y)
{

    lv_obj_t *grad = lv_obj_create(obj);
    lv_obj_set_width(grad, 687);
    lv_obj_set_height(grad, 50);
    lv_obj_set_x(grad, 0);
    lv_obj_set_y(grad, y);
    lv_obj_set_align(grad, LV_ALIGN_TOP_LEFT);
    lv_obj_clear_flag(grad, LV_OBJ_FLAG_SCROLLABLE); /// Flags
    lv_obj_set_style_bg_color(grad, lv_color_hex(color), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(grad, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_grad_color(grad, lv_color_hex(0x000000), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_main_stop(grad, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_grad_stop(grad, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_grad_dir(grad, LV_GRAD_DIR_HOR, LV_PART_MAIN | LV_STATE_DEFAULT);
    return grad;
}

void OnAddOneClicked(lv_event_t *e)
{
    static uint8_t cnt = 0;
    cnt++;
    lv_label_set_text_fmt(ui_lblCountValue, "%d", cnt);
}

void OnRotateClicked(lv_event_t *e)
{
    auto disp = lv_disp_get_default();
    auto rotation = (lv_disp_rot_t)((lv_disp_get_rotation(disp) + 1) % (LV_DISP_ROT_270 + 1));
    lv_disp_set_rotation(disp, rotation);
}

void setup()
{
#ifdef ARDUINO_USB_CDC_ON_BOOT
    delay(5000);
#endif
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    log_i("Board: %s", BOARD_NAME);
    log_i("CPU: %s rev%d, CPU Freq: %d Mhz, %d core(s)", ESP.getChipModel(), ESP.getChipRevision(), getCpuFrequencyMhz(), ESP.getChipCores());
    log_i("Free heap: %d bytes", ESP.getFreeHeap());
    log_i("Free PSRAM: %d bytes", ESP.getPsramSize());
    log_i("SDK version: %s", ESP.getSdkVersion());

    smartdisplay_init();

    __attribute__((unused)) auto disp = lv_disp_get_default();
    // lv_disp_set_rotation(disp, LV_DISP_ROT_90);
    // lv_disp_set_rotation(disp, LV_DISP_ROT_180);
    // lv_disp_set_rotation(disp, LV_DISP_ROT_270);

    ui_init();

    // To use third party libraries, enable the define in lv_conf.h: #define LV_USE_QRCODE 1
    auto ui_qrcode = lv_qrcode_create(ui_scrMain, 100, lv_color_black(), lv_color_white());
    const char *qr_data = "https://github.com/rzeldent/esp32-smartdisplay";
    lv_qrcode_update(ui_qrcode, qr_data, strlen(qr_data));
    lv_obj_center(ui_qrcode);

    createGradient(ui_scrMain, 0xFF0000, 50);
    createGradient(ui_scrMain, 0x00FF00, 100);
    createGradient(ui_scrMain, 0x0000FF, 150);
    createGradient(ui_scrMain, 0xFFFF00, 200);
    createGradient(ui_scrMain, 0xFF00FF, 250);
}

ulong next_millis;

void loop()
{
    auto const now = millis();
    if (now > next_millis)
    {
        next_millis = now + 500;

        char text_buffer[32];
        sprintf(text_buffer, "%lu", now);
        lv_label_set_text(ui_lblMillisecondsValue, text_buffer);

#ifdef BOARD_HAS_RGB_LED
        auto const rgb = (now / 2000) % 8;
        smartdisplay_led_set_rgb(rgb & 0x01, rgb & 0x02, rgb & 0x04);
#endif

#ifdef BOARD_HAS_CDS
        auto cdr = analogReadMilliVolts(CDS);
        sprintf(text_buffer, "%d", cdr);
        lv_label_set_text(ui_lblCdrValue, text_buffer);
#endif
    }

    lv_timer_handler();
}```

xperiments, 2024-07-12:
Found the culpit

FROM
"'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=12'",

"'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=21'",

xperiments, 2024-07-12:
After the change
[image]

rzeldent, 2024-07-13:
Hi Predro,

What a good find! You already made a PR, nice! Will commit it shortly..

Thanks,

Rene

closed 2024-07-13 by xperiments