rzeldent/esp32-smartdisplay #197: Add

paaspaas00, 2024-09-05:
You have some nicely-defined boards here. However, people may want to use the library for user with a compatible display (for which there are drivers in this lib) with this library's touch drivers etc. 
To do so people are currently forced to write a custom board definition.
To make stuff easier, would it be possible to provide a nice programming API exposing basically everything it's exposed via compile flags, bit as C/C++ functions? So people can mix and match display, touch etc straight from code, without writing a board.
The board config for this would simply be "esp32_custom"

paaspaas00, 2024-09-05:
basically something like #126

rzeldent, 2024-09-07:
You can also remove the git-submodule and create your own board (or start with a predefined board) in the boards directory...
This is already possible...

paaspaas00, 2024-09-07:
@rzeldent hi, thanks for your reply! Yes I know it's already possible with custom boards, but I meant more of a way to define stuff via software, in C++. Like create a display with some constructor, create a touchscreen with some constructor etc

rzeldent, 2024-09-08:
The driver is now written in C, not C++. This would involve quite some changes.
Somewhere in the future this would be something to make classes out of the drivers but for now not a priority...

paaspaas00, 2024-09-08:
@rzeldent thank you! I mean, C++ is just easier with classes etc, but no problem, the API could be made available in C as well, no big deal. You could as well expose C APIs