rzeldent/esp32-smartdisplay #198: 8048S043 capacitative touch buttons not working in demo

rzeldent, 2024-09-08:
### Discussed in https://github.com/rzeldent/esp32-smartdisplay/discussions/189

<div type='discussions-op-text'>

<sup>Originally posted by **ralphrmartin** August 14, 2024</sup>
I have a problem on an 8048s043 board, bought as having capacitative touch. It has both a GT911 chip and an xpt2046 chip.
The board as supplied had a demo running on it iwhich iworked OK, including touch (which does indeed seem to be capacitative as it responded to a light touch), but when I try the eap32-smartdisplay-demo program, touch does not work correctly (whether I select 8048S043C or 8048S043R). The display is drawn correctly, but the buttons to add 1 and to rotate do not function.

On enabling debugging, the display" jumps" whenever I press either button, but they still do not work as intended. The debugging information shows the below. 

Can anyone enlighten me as to what is going wrong here?
(Why is the SDK version: v4.4.7-dirty?)

Thanks.

```

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   139][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  5149][I][main.cpp:29] setup(): Board: esp32-8048S043C
[  5155][I][main.cpp:30] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s)
[  5162][I][main.cpp:31] setup(): Free heap: 316116 bytes
[  5167][I][main.cpp:32] setup(): Free PSRAM: 8386279 bytes
[  5173][I][main.cpp:33] setup(): SDK version: v4.4.7-dirty
[  5178][D][esp32_smartdisplay.c:173] smartdisplay_init(): smartdisplay_init
[  5185][D][lvgl_panel_st7262_par.c:65] lvgl_lcd_init(): rgb_panel_config: clk_src:0, timings:{pclk_hz:12500000, h_res:800, v_res:480, hsync_pulse_width:4, hsync_back_porch:8, hsync_front_porch:8, vsync_pulse_width:4, vsync_back_porch:8, vsync_front_porch:8, flags:{hsync_idle_low:0, vsync_idle_low:0, de_idle_high:0, pclk_active_neg:1, pclk_idle_high:0}}, data_width:16, sram_trans_align:4, psram_trans_align:64, hsync_gpio_num:39, vsync_gpio_num:41, de_gpio_num:40, pclk_gpio_num:42, data_gpio_nums:[15,16,4,45,48,47,21,14,8,3,46,9,1,5,6,7,], disp_gpio_num:-1, on_frame_trans_done:0x42029c64, user_ctx:0x3fca4718, flags:{disp_active_low:0, relax_on_idle:0, fb_in_psram:1}
[  5244][D][lvgl_panel_st7262_par.c:66] lvgl_lcd_init(): refresh rate: 0 Hz
[  5275][D][lvgl_touch_gt911_i2c.c:45] lvgl_touch_init(): i2c_config: mode:1, sda_io_num:19, scl_io_num:20, sda_pullup_en:0, scl_pullup_en:0, master:{clk_speed:400000}, clk_flags:0
[  5292][D][lvgl_touch_gt911_i2c.c:60] lvgl_touch_init(): io_i2c_config: dev_addr:0x5d, control_phase_bytes:1, user_ctx:0x3fca46f8, dc_bit_offset:0, lcd_cmd_bits:16, lcd_param_bits:0, flags:{.dc_low_on_data:0, disable_control_phase:1}
[  5312][D][lvgl_touch_gt911_i2c.c:75] lvgl_touch_init(): touch_config: x_max:800, y_max:480, rst_gpio_num:38, int_gpio_num:18, levels:{reset:0, interrupt:0}, flags:{swap_xy:0, mirror_x:0, mirror_y:0}, user_data:0x3fcec7d4
[  5384][D][esp_touch_gt911.c:147] gt911_read_info(): GT911 productId: 911
[  5390][D][esp_touch_gt911.c:148] gt911_read_info(): GT911 fwId: 0x1060
[  5397][D][esp_touch_gt911.c:149] gt911_read_info(): GT911 xResolution/yResolution: (0,0)
[  5405][D][esp_touch_gt911.c:150] gt911_read_info(): GT911 vendorId: 0xff
[  5412][D][esp_touch_gt911.c:465] esp_lcd_touch_new_i2c_gt911(): handle:0x3d977b08
[ 35956][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(421,45), area:53
[ 35965][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:421, y:45, strength:53
[ 35986][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(421,45), area:53
[ 35995][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:421, y:45, strength:53
[ 36016][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(421,45), area:53
[ 36025][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:421, y:45, strength:53
[ 36046][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(421,45), area:53
[ 36055][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:421, y:45, strength:53
[ 36076][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(421,45), area:53
[ 36085][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:421, y:45, strength:53
[ 36106][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(421,45), area:53
[ 36115][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:421, y:45, strength:53
[ 36646][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(413,52), area:70
[ 36655][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:413, y:52, strength:70
[ 36676][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(413,52), area:70
[ 36685][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:413, y:52, strength:70
[ 36706][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(413,52), area:70
[ 36715][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:413, y:52, strength:70
[ 36736][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(413,52), area:70
[ 36745][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:413, y:52, strength:70
[ 36766][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(413,52), area:70
[ 36775][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:413, y:52, strength:70
[ 36796][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(413,52), area:70
[ 36805][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:413, y:52, strength:70
[ 36826][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(413,52), area:70
[ 36835][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:413, y:52, strength:70
[ 36856][D][esp_touch_gt911.c:266] gt911_read_data(): Point: #0, event:0, point:(413,52), area:64
[ 36865][D][esp_touch_gt911.c:303] gt911_get_xy(): Touch data: x:413, y:52, strength:64
```</div>

DevAnyKR, 2025-01-18:
i have esp32-8048s043C  no xpt2046, only gt911
ST7262 RGB

```
'.pio/libdeps/esp32-8048S043C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:101:10: error: 'esp_lcd_rgb_panel_config_t' has no member named 'on_frame_trans_done'
.pio/libdeps/esp32-8048S043C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:102:10: error: 'esp_lcd_rgb_panel_config_t' has no member named 'user_ctx'
```

where is on_frame_trans_done, user_ctx ?

ps. I also have the screen jumping problem.

ex) LVGL
```
lv_display_set_buffers(disp, disp_draw_buf, NULL, bufSize * 2, LV_DISPLAY_RENDER_MODE_PARTIAL); // is OK.

lv_display_set_buffers(disp, disp_draw_buf, NULL, bufSize * 2, LV_DISPLAY_RENDER_MODE_DIRECT); // screen moves every time touch 
```

rzeldent, 2025-01-18:
Hi DevAny,

Strange it does not compile. Are you using the coorrect version of LVGL? Otherwise take a look at the develop branch.

I only uses partial updates. It might be possible that complete updates work but the framerate will be very low and this can lead to problems. It can also be that the parameters PCLK etc need to be tweaked.

Hope I gave you some useful info ...

avbazurt, 2025-08-07:
First of all, thank you for this amazing project. I recently started working with TFT displays, and this library has helped me a lot. Everything worked great with non-touch displays (like the ESP32-4827S043N).

[image]

[image]

However, when I tried the ESP32-8048S043C and ran the example from the repository, the screen stays completely blank. The touch seems to respond (according to the logs with -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG), but nothing is displayed.

[image]

I tested with the latest commits:

- esp32-smartdisplay-demo: d031a0a
- esp32-smartdisplay: 6d6be99
- platformio-espressif32-sunton: 3cf74ce

And also with the latest releases:

- demo: 2.1.1
- smartdisplay: 2.1.0
- sunton: 2.0.11

But I get the same result. Maybe I’m missing something. I’d really appreciate any guidance to get this display working with this great library.