rzeldent/esp32-smartdisplay #208: lvgl 9.2.1 breaks develop branch

twiesenthal, 2024-10-25:
Hi,

i'm fairly new to esp32 programming and using [esp32-smartdisplay](https://github.com/rzeldent/esp32-smartdisplay) in general, so i don't feel confident enough to provide a fox for this in a merge request.
But i would like to let you know that with the merge from [this issue](https://github.com/lvgl/lvgl/issues/6819), which was included in lvgl 9.2.1 yesterday, the develop branch is currently broken 

sw_rotate was removed which results in build time errors like this. 

```
esp32_smartdisplay/src/esp32_smartdisplay.c:197:15: error: 
'lv_display_t' {aka 'struct lv_display_t'} has no member named 'sw_rotate'
```

The simple solution is of course to pin the lvgl version to 9.2.0 in `library.json`, but that does not seem sustainable to me. Maybe someone more confident with the code can fix the root cause if the error.

rzeldent, 2024-10-30:
Just committed fix for rotation. https://github.com/rzeldent/esp32-smartdisplay/tree/lvgl9.2

Does this work for you?

bobskigit, 2024-10-30:
it works for me

rzeldent, 2024-10-30:
Retested on software rotation device esp32-4848S040CIY1. Worked as expected.

PR to have this in the develop branch.
https://github.com/rzeldent/esp32-smartdisplay/pull/210

twiesenthal, 2024-11-01:
Thank you very much @rzeldent. Works like a charm

closed 2024-11-01 by twiesenthal