rzeldent/esp32-smartdisplay #217: Support for JC3248W535EN

NorthernMan54, 2024-11-08:
Would it be possible to add support for this board [JC3248W535EN](https://s.click.aliexpress.com/e/_DFO5uIV?)

The display and touch interface is based on the AXS15231B chip, which isn't currently supported, but the vendor supplied an [ESP-IDF](https://github.com/NorthernMan54/JC3248W535EN) based build.

NorthernMan54, 2024-11-08:
PS - And an Android build is in the ArduinoLvgl branch

JoDaNl, 2024-11-11:
This is just such a coincidence...I was ready to open an issue on the same topic this evening.

A few weeks ago I bought exactly this display for an existing project. As my project supports a few types of these CYD's I was using the esp32-smartdisplay library.

For initial testing of this display I used the [Arduino_GFX ](https://github.com/moononournation/Arduino_GFX)library. This worked fine. However I had the need of a single library supporting different types/brands of CYDs'. 

So I tried to integrate support for the AXS15231B driver-chip to the esp32-library. 
Basis for my work was information I found here:
- [espressif/esp_lcd_axs15231b](https://components.espressif.com/components/espressif/esp_lcd_axs15231b)
- [Spotpear](https://spotpear.com/wiki/ESP32-S3-3.5-inch-LCD-Captive-TouchScreen-Display-480x320-Tablet-MP3-Video-Weather-Clock.html)

So far I managed to integrate the expressif code into my development branch and have a first image on the display, using the [esp32-smartdisplay-demo](https://github.com/rzeldent/esp32-smartdisplay-demo). So that is a first step. See images 

[image]
[image]
 
Unfortunately I'm bumping into the following issue:

The display will do the 1st display refresh correctly. However subsequent display updates are rendered wrong. The 'microseconds' value should update twice every second. Unfortunately it doesn't. The result after 5 updates is shown below.

[image]

My debugging shows that partial updates for this chip are not working.
If there's interest I can share code.

JoDaNl, 2024-11-12:
Alright. I created a single repo which

- has the esp32-smartdisplay-demo code in the src foler
- has a new board entry in the boards folder (esp32-JC3248W535.json)
- contains the development-code for the esp32-smartdisplay library in the lib-folder (so for this dev-work not making usinf of .gitmodules

Best to cleanup/delete the %user%\.platformio folder as I'm using a custom IDF5 based platform. Otherwise it will not compile !

Once compiles & uploaded the display will start and show the demo. The terminal will print the coordinates of display-updates.
The first 8 are related to the initial full-screen update (as the display-buffer is 1/8 of the screen-size).

Subsequent updates are from the 'milliseconds'  label. As of update 115 onwards you will see some corrupted pixels.
I did quite some debuggint to unserstand what is going on. My conclusion so far is that this command...

`        tx_color(axs15231b, io, LCD_CMD_RAMWR, color_data, len);`  (esp_panel_axs15231b.c line 383)

... always starts writing at pixel (0,0) regardless of the RASET & CASET registers

NorthernMan54, 2024-11-13:
Tks for sharing, I looked at your code, and compared it to the provided examples, and even used the same demo.  One thing I did find was that the vendor examples all do full screen refresh, and when turning on partial, they all had the same pixellation issue.

I'm wondering if the register documentation is incorrect or the hardware has an issue / limitation here

NorthernMan54, 2024-11-13:
Looking further and my guess is that the code never implemented partial refresh, as the existing logic is broken for partial. I looked at the data sheet versus what is being sent, and it makes no sense for partial.

when sending partial, you need to set CASET and RASET, then send with RAMWR.  But it does this, CASET, then RAMWRC, which is just wrong. Unless I read the data incorrectly

NorthernMan54, 2024-11-14:
And I tried that, no luck.  Would the best course be, if you really need partial refresh be to working,  to raise an issue against the actual driver - https://components.espressif.com/components/espressif/esp_lcd_axs15231b/versions/1.0.0?language=en

JoDaNl, 2024-11-18:
I did some further testing & investigations. Also checked other implementations (like ESPHome driver & Arduino-GFX library).
No implementation so far works with partial refresh.

My conclusion is that there's some functionality missing in the driver chip (bug).

Setting the CASET & RASET registers DOES allow me to create a window, however it is NOT POSSIBLE to position this window, i.e. the window is always in the upper-left corner.

I have moves to full refresh and splitted to update of the screen in multiple parts due to a DMA issue. Example has been committed. Would like to move aways from the  cursom IDF5 based platform defition 

```
PLATFORMIO.INI
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5

```
Howers setting platform back to 
[platform = espressif32](url)

Is not working yet due to QSPI changes in ID4.4 versus IDF5.x

rzeldent, 2024-12-27:
Made an implementation for the JC3248W535N in the current develop. Do not own the board and still work in progress...
@JoDaNl You made good progress and think that a cleaner driver might resolve the issue. Maybe this addition solves the issues?
QSPI indeed is not yet possible; a pity...

Shadowtrance, 2025-01-01:
Just tried out the develop branch with mine and it's a no go (with no changes by me).
Screen boots up, full white, then slowly bleeds in from the edges until the whole screen is black (backlight still on) and then no change.
I will tinker a bit and see what i can figure out. :)

Shadowtrance, 2025-01-05:
No progress so far :( And I honestly don't know why.
Looking at the related files you've done vs the files that are in the zip from the vendor they're not really that different.
Main difference is they have the lv_port thing along with the bsp / screen / touch files vs how you've done it with the new esp_kcd_xxx. So different approaches but basically the same thing. I am confused. lol Oh and lvgl 9.2.2 vs max lvgl 8.3.11.
So yeah so far only their files and lvgl 8.3.11 are running for me.
We'll figure it out! 🤣