rzeldent/esp32-smartdisplay #219: Doesn't build on the latest espressif32 (6.9.0)

yanxke, 2024-11-28:
Need to set in platformio.ini

```
[env]
platform = espressif32@6.8.1
```

to build.

rzeldent, 2024-12-26:
HI yanxke,

Just retested without modifications and works. Did you updated platformIO?

Removing unused dependencies...
Library Manager: Removing test @ 0.0.0+20241107223720
Library Manager: test@0.0.0+20241107223720 has been removed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-1732S019C.html
PLATFORM: Espressif 32 (6.9.0) > esp32-1732S019C
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing file://C:\Users\rzeld\source\repos\esp32-smartdisplay\test
Library Manager: test@0.0.0+20241226214310 has been installed!
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 9.2.2
|-- test @ 0.0.0+20241226214310
Building in release mode
Compiling .pio\build\esp32-1732S019C\src\esp32_smartdisplay.c.o
Compiling .pio\build\esp32-1732S019C\src\esp_io_expander.c.o
Compiling .pio\build\esp32-1732S019C\src\esp_lcd_panel_io_3wire_spi.c.o
Compiling .pio\build\esp32-1732S019C\src\esp_lcd_touch.c.o
...
...
Compiling .pio\build\esp32-1732S019C\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\esp32-1732S019C\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32-1732S019C\libFrameworkArduino.a
Linking .pio\build\esp32-1732S019C\firmware.elf
Retrieving maximum program size .pio\build\esp32-1732S019C\firmware.elf
Checking size .pio\build\esp32-1732S019C\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.6% (used 18212 bytes from 327680 bytes)
Flash: [          ]   3.6% (used 237309 bytes from 6553600 bytes)
Building .pio\build\esp32-1732S019C\firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
========================================================================================================================= [SUCCESS] Took 54.29 seconds =========================================================================================================================

Environment      Status    Duration
---------------  --------  ------------
esp32-1732S019C  SUCCESS   00:00:54.286
========================================================================================================================= 1 succeeded in 00:00:54.286 ========================================================================================================================= 
 *  Terminal will be reused by tasks, press any key to close it.

yanxke, 2025-01-15:
Yes, confirmed to work now.  Thanks!

closed 2025-01-15 by yanxke