rzeldent/esp32-smartdisplay #221: esp32-8048S050C Wrong Config

xperiments, 2024-11-30:
Hi @rzeldent,

I have a customer my new 5inch version was not working, screen was white....

After some research (and some fight about speed w Sunton guy who helped a lot)  I have found the bug to be in the board config

```
      "'-D ST7262_PANEL_CONFIG_CLK_SRC=LCD_CLK_SRC_PLL160M'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(12.5*1000000)'",
```

      to

```
      "'-D ST7262_PANEL_CONFIG_CLK_SRC=LCD_CLK_SRC_PLL240M'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(18*1000000)'",
```      

Tested timings up to 20 (artifacts at screen)

Tested from 16M to 19M with good results

Cheers!!!

rzeldent, 2024-12-02:
Hi Pedro,

I do not own this board so cannot validate it works. However I changed the PCLK to 16Mhz in the **develop** branch of the:
- esp32-8048S050C
- esp32-8048S050R
- esp32-8048S050N

I also tested other bords like the esp32-4848S040CIY1, esp32-8048S070C to use 18Mhz but they start flickering, so reverted that
I hope this works...

xperiments, 2024-12-02:
@rzeldent The most needed change is the LCD_CLK_SRC_PLL240M

Is that what make the change from not working to working

BTW I have 7, 5 and 2.8 board if you need any test

xperiments, 2024-12-02:
Actually w this change it will work fine until 16M

To go up to 24M ( the screen spec) we need to activate 120M in PSRAM but that is only avaliable on idf 5x

xperiments, 2024-12-02:
@rzeldent can be great if you can come to discord here https://discord.gg/yRNWJj55 thx!!

rzeldent, 2024-12-26:
Invite is expired...