rzeldent/esp32-smartdisplay #226: Errors when initialising esp32-s3touchlcd7

bobskigit, 2025-01-08:
I get these errors when initializing esp32-s3touchlcd7. GUI and touch work ok. Any ideas?

[  5272][I][main.cpp:107] setup(): Board: esp32-s3touchlcd7
[  5278][I][main.cpp:108] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s)
[  5285][I][main.cpp:109] setup(): Free heap: 262180 bytes
[  5291][I][main.cpp:110] setup(): Free PSRAM: 8386279 bytes
[  5296][I][main.cpp:111] setup(): SDK version: v4.4.7-dirty
[  5302][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (5057) gpio: gpio_set_level(227): GPIO output gpio_num error
E (5057) ledc: ledc_channel_config(608): gpio_num argument is invalid
[  5335][W][esp_touch_gt911.c:87] gt911_reset(): No RST pin defined

Sanjaypatel145, 2025-02-11:
same problem here.
Did you find solution?

bobskigit, 2025-02-11:
No it works fine, just ignored the errors

Sanjaypatel145, 2025-03-02:
fine