rzeldent/esp32-smartdisplay #232: esp32-1732S019N Rotation 90/270 Broken

lincomatic, 2025-02-02:
On the 1732S019N (ST7789 170x320), rotation 0 and 180 work properly, but 90 and 270 result in garbage thin horizontal dotted lines on the screen.

I'm using the lines from your sample code:
```
    smartdisplay_init();

    //__attribute__((unused)) auto disp = lv_disp_get_default();
    lv_disp_set_rotation(disp, LV_DISP_ROTATION_90);
    // lv_disp_set_rotation(disp, LV_DISP_ROTATION_180);
    // lv_disp_set_rotation(disp, LV_DISP_ROTATION_270);

    ui_init();
```

rzeldent, 2025-02-08:
Are you using the demo?
It the dimensions might be out of bounds and this might result in writing outside the buffer (this creating strange images)...

GeoffWebster, 2025-07-03:
Any progress on this as I have the same problem. Workaround from antlanc7 didn't fix it for me.

GeoffWebster, 2025-07-04:
antlanc7' workaround didn't work for me (see previous post) but using his workaround as a clue, I came up with working display for 90 deg and 270 deg by using "'-D DISPLAY_GAP_X=-20'",       "'-D DISPLAY_GAP_Y=35'".

antlanc7, 2025-07-04:
Yeah it depends on the specific display you have.

GeoffWebster, 2025-07-04:
These values work with the built-in display (170 x 240) of the ESP32-1732S019N board, the same one that started the ball rolling on this bug report