rzeldent/esp32-smartdisplay #238: how to create new board json config pin of ESP32-S3-Touch-LCD-4.3B?

winitboonkliang, 2025-02-09:
my new board from waveshare  ESP32-S3-Touch-LCD-4.3B
https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3B

LCD ST7262 

this demo code config pin from waveshare

#define I2C_MASTER_SCL_IO           9       /*!< GPIO number used for I2C master clock */
#define I2C_MASTER_SDA_IO           8       /*!< GPIO number used for I2C master data  */
#define I2C_MASTER_NUM              0       /*!< I2C master i2c port number, the number of i2c peripheral interfaces available will depend on the chip */
#define I2C_MASTER_FREQ_HZ          400000                     /*!< I2C master clock frequency */
#define I2C_MASTER_TX_BUF_DISABLE   0                          /*!< I2C master doesn't need buffer */
#define I2C_MASTER_RX_BUF_DISABLE   0                          /*!< I2C master doesn't need buffer */
#define I2C_MASTER_TIMEOUT_MS       1000

#define GPIO_INPUT_IO_4    4
#define GPIO_INPUT_PIN_SEL  1ULL<<GPIO_INPUT_IO_4
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////// Please update the following configuration according to your LCD spec //////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define EXAMPLE_LCD_H_RES               (LVGL_PORT_H_RES)
#define EXAMPLE_LCD_V_RES               (LVGL_PORT_V_RES)
#define EXAMPLE_LCD_PIXEL_CLOCK_HZ      (16 * 1000 * 1000)
#define EXAMPLE_LCD_BIT_PER_PIXEL       (16)
#define EXAMPLE_RGB_BIT_PER_PIXEL       (16)
#define EXAMPLE_RGB_DATA_WIDTH          (16)
#define EXAMPLE_RGB_BOUNCE_BUFFER_SIZE  (EXAMPLE_LCD_H_RES * CONFIG_EXAMPLE_LCD_RGB_BOUNCE_BUFFER_HEIGHT)
#define EXAMPLE_LCD_IO_RGB_DISP         (-1)             // -1 if not used
#define EXAMPLE_LCD_IO_RGB_VSYNC        (GPIO_NUM_3)
#define EXAMPLE_LCD_IO_RGB_HSYNC        (GPIO_NUM_46)
#define EXAMPLE_LCD_IO_RGB_DE           (GPIO_NUM_5)
#define EXAMPLE_LCD_IO_RGB_PCLK         (GPIO_NUM_7)
#define EXAMPLE_LCD_IO_RGB_DATA0        (GPIO_NUM_14)
#define EXAMPLE_LCD_IO_RGB_DATA1        (GPIO_NUM_38)
#define EXAMPLE_LCD_IO_RGB_DATA2        (GPIO_NUM_18)
#define EXAMPLE_LCD_IO_RGB_DATA3        (GPIO_NUM_17)
#define EXAMPLE_LCD_IO_RGB_DATA4        (GPIO_NUM_10)
#define EXAMPLE_LCD_IO_RGB_DATA5        (GPIO_NUM_39)
#define EXAMPLE_LCD_IO_RGB_DATA6        (GPIO_NUM_0)
#define EXAMPLE_LCD_IO_RGB_DATA7        (GPIO_NUM_45)
#define EXAMPLE_LCD_IO_RGB_DATA8        (GPIO_NUM_48)
#define EXAMPLE_LCD_IO_RGB_DATA9        (GPIO_NUM_47)
#define EXAMPLE_LCD_IO_RGB_DATA10       (GPIO_NUM_21)
#define EXAMPLE_LCD_IO_RGB_DATA11       (GPIO_NUM_1)
#define EXAMPLE_LCD_IO_RGB_DATA12       (GPIO_NUM_2)
#define EXAMPLE_LCD_IO_RGB_DATA13       (GPIO_NUM_42)
#define EXAMPLE_LCD_IO_RGB_DATA14       (GPIO_NUM_41)
#define EXAMPLE_LCD_IO_RGB_DATA15       (GPIO_NUM_40)

#define EXAMPLE_LCD_IO_RST              (-1)             // -1 if not used
#define EXAMPLE_PIN_NUM_BK_LIGHT        (-1)    // -1 if not used
#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL   (1)
#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL  !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL

#define EXAMPLE_PIN_NUM_TOUCH_RST       (-1)            // -1 if not used
#define EXAMPLE_PIN_NUM_TOUCH_INT       (-1)            // -1 if not used

### config esp32-8048S043C.json file is RGB pin  but demo code form waveshare pin is DATA0-DATA16

      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R0=8'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R1=3'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R2=46'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R3=9'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R4=1'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G0=5'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G1=6'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G2=7'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G3=15'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G4=16'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G5=4'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B0=45'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B1=48'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B2=47'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=21'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B4=14'",

rzeldent, 2025-02-09:
Hi winitboonkliang,

You just have to put these values in the same format as the existing board defines. Then make sure it is in the boards directory.
If it is working I can add it to the repository of the defines so it will be available to everyone...

winitboonkliang, 2025-02-09:
Touch screen ok
LCD color incorrect

      "'-D ARDUINO_ESP32S3_DEV'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=0'",
      "'-D ESP32_8048S043X'",
      "'-D DISPLAY_WIDTH=800'",
      "'-D DISPLAY_HEIGHT=480'",
      "'-D DISPLAY_SOFTWARE_ROTATION'",
      "'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT)'",
      "'-D LVGL_BUFFER_MALLOC_FLAGS=(MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT)'",
      "'-D GPIO_BCKL=-1'",
      "'-D DISPLAY_ST7262_PAR'",
      "'-D ST7262_PANEL_CONFIG_CLK_SRC=LCD_CLK_SRC_PLL160M'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(12.5*1000000)'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_H_RES=DISPLAY_WIDTH'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_V_RES=DISPLAY_HEIGHT'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_PULSE_WIDTH=4'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_BACK_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_FRONT_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_PULSE_WIDTH=4'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_BACK_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_FRONT_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_HSYNC_IDLE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_VSYNC_IDLE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_DE_IDLE_HIGH=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_ACTIVE_NEG=true'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_IDLE_HIGH=false'",
      "'-D ST7262_PANEL_CONFIG_DATA_WIDTH=16'",
      "'-D ST7262_PANEL_CONFIG_SRAM_TRANS_ALIGN=4'",
      "'-D ST7262_PANEL_CONFIG_PSRAM_TRANS_ALIGN=64'",
      "'-D ST7262_PANEL_CONFIG_HSYNC_GPIO_NUM=46'",
      "'-D ST7262_PANEL_CONFIG_VSYNC_GPIO_NUM=3'",
      "'-D ST7262_PANEL_CONFIG_DE_GPIO_NUM=5'",
      "'-D ST7262_PANEL_CONFIG_PCLK_GPIO_NUM=7'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R0=14'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R1=38'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R2=18'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R3=17'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R4=10'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G0=39'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G1=0'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G2=45'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G3=48'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G4=47'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G5=21'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B0=1'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B1=2'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B2=42'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=41'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B4=40'",
      "'-D ST7262_PANEL_CONFIG_DISP_GPIO_NUM=GPIO_NUM_NC'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_DISP_ACTIVE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_RELAX_ON_IDLE=false'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_FB_IN_PSRAM=true'",
      "'-D DISPLAY_SWAP_XY=false'",
      "'-D DISPLAY_MIRROR_X=false'",
      "'-D DISPLAY_MIRROR_Y=false'",
      "'-D BOARD_HAS_TOUCH'",
      "'-D CONFIG_ESP_LCD_TOUCH_MAX_POINTS=1'",
      "'-D CONFIG_ESP_LCD_TOUCH_MAX_BUTTONS=0'",
      "'-D TOUCH_GT911_I2C'",
      "'-D GT911_I2C_HOST=I2C_NUM_0'",
      "'-D GT911_I2C_CONFIG_SDA_IO_NUM=8'",
      "'-D GT911_I2C_CONFIG_SCL_IO_NUM=9'",
      "'-D GT911_I2C_CONFIG_SDA_PULLUP_EN=GPIO_PULLUP_DISABLE'",
      "'-D GT911_I2C_CONFIG_SCL_PULLUP_EN=GPIO_PULLUP_DISABLE'",
      "'-D GT911_I2C_CONFIG_MASTER_CLK_SPEED=400000'",
      "'-D GT911_I2C_CONFIG_CLK_FLAGS=I2C_SCLK_SRC_FLAG_FOR_NOMAL'",
      "'-D GT911_IO_I2C_CONFIG_DEV_ADDR=GT911_IO_I2C_CONFIG_DEV_ADDRESS_5D'",
      "'-D GT911_IO_I2C_CONFIG_CONTROL_PHASE_BYTES=1'",
      "'-D GT911_IO_I2C_CONFIG_DC_BIT_OFFSET=0'",
      "'-D GT911_IO_I2C_CONFIG_LCD_CMD_BITS=16'",
      "'-D GT911_IO_I2C_CONFIG_LCD_PARAM_BITS=0'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DC_LOW_ON_DATA=false'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DISABLE_CONTROL_PHASE=true'",
      "'-D GT911_TOUCH_CONFIG_X_MAX=DISPLAY_WIDTH'",
      "'-D GT911_TOUCH_CONFIG_Y_MAX=DISPLAY_HEIGHT'",
      "'-D GT911_TOUCH_CONFIG_RST_GPIO_NUM=-1'",
      "'-D GT911_TOUCH_CONFIG_INT_GPIO_NUM=4'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_RESET=-1'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_INTERRUPT=0'",
      "'-D TOUCH_SWAP_XY=false'",
      "'-D TOUCH_MIRROR_X=false'",
      "'-D TOUCH_MIRROR_Y=false'",
      "'-D BOARD_HAS_TF=true'",
      "'-D TF_CS=10'",
      "'-D TF_SPI_MOSI=11'",
      "'-D TF_SPI_SCLK=12'",
      "'-D TF_SPI_MISO=13'"

[image]

winitboonkliang, 2025-02-09:
this color correct

[image]

winitboonkliang, 2025-02-09:
st7262 library file part? for edit some pin used extra port RST  used EXIO3 
this board uesd CH422G for IOExpander

gavzzorz, 2025-03-04:
Hi Winitboonkliang,

what is the display in your last image posted? im trying to implement this library wis a ESP32-S3-Touch-LCD-4 and im having issue with colours displaying correctly.

Did you manage to resolve your issue?

closed 2026-10-10 by github-actions[bot]