rzeldent/esp32-smartdisplay #244: Double buffering

gopro2027, 2025-02-17:
I'm trying to modify `lvgl_panel_st7789_spi.c` to use double buffering via `esp_lcd_rgb_panel_get_frame_buffer(panel_handle, 2, &buf1, &buf2);` which i saw in another issue:
https://github.com/rzeldent/esp32-smartdisplay/issues/199
The function doesn't seem to exist and frankly im just sure why.
I've also tried not using that function and just allocating a second buffer instead and putting that in place of null and changing render mode to direct:
```
void *drawBuffer = heap_caps_malloc(drawBufferSize, LVGL_BUFFER_MALLOC_FLAGS);
void *drawBuffer2 = heap_caps_malloc(drawBufferSize, LVGL_BUFFER_MALLOC_FLAGS);
lv_display_set_buffers(display, drawBuffer, drawBuffer2, drawBufferSize, LV_DISPLAY_RENDER_MODE_DIRECT);
```
This does not seem to work too. Black screen.

hillsandales, 2025-03-23:
I'm getting a black screen as well. I'm finding that the above heap_caps_malloc() function is causing problems. It seems the memory is not being allocated. I have -DCONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS set. When it gets to this function, it aborts, and prints `Mem alloc fail. size 0x0000bb80 caps 0x00000404` to the terminal.

I have the following defines set. lv_conf.h LV_MEM_SIZE is set to default of 64 kB. I've tried increasing this for the display size, but no luck. 
```
  "LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT/16)"
  "LVGL_BUFFER_MALLOC_FLAGS=(MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT)"
```

With these defines, I'm trying to allocate a drawBufferSize of 48000 bytes, so I would think it should work. The heap_caps_get_*() functions all return zero. xPortgetFreeHeapSize() returns 279292.

Been trying to figure this out for hours and still not there...

Note: I started using platformio configs and got the screen working, but needing some other libraries, I switched over to esp-idf build system, so I'm also learning that, and not sure if that's causing my issues allocating memory...

gopro2027, 2025-03-24:
> I'm getting a black screen as well. I'm finding that the above heap_caps_malloc() function is causing problems. It seems the memory is not being allocated. I have [...]

So wait, were you able to get double buffering working? What configs did you use? Also I'm still using basic platformio stuff, not esp-idf.

hillsandales, 2025-03-28:
No I haven't tried double buffering.

closed 2026-10-09 by github-actions[bot]