rzeldent/esp32-smartdisplay #246: Display Colour Issue

gavzzorz, 2025-02-24:
Hi All, hoping someone can help with an issue im having with a WaveShare ESP32-S3-Touch-LCD-4.

I have created the following Boards File...
{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "'-D ARDUINO_ESP32S3_DEV'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=0'",
      "'-D ESP32_S3TOUCHLCD4'",
      "'-D DISPLAY_WIDTH=480'",
      "'-D DISPLAY_HEIGHT=480'",
      "'-D DISPLAY_SOFTWARE_ROTATION'",
      "'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT)'",
      "'-D LVGL_BUFFER_MALLOC_FLAGS=(MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT)'",
      "'-D GPIO_BCKL=-1'",
      "'-D DISPLAY_ST7701_PAR'", 
      "'-D ST7701_IO_3WIRE_SPI_LINE_CONFIG_CS_GPIO_NUM=42'",
      "'-D ST7701_IO_3WIRE_SPI_LINE_CONFIG_SCL_GPIO_NUM=2'",
      "'-D ST7701_IO_3WIRE_SPI_LINE_CONFIG_SDA_GPIO_NUM=1'",
      "'-D ST7701_IO_3WIRE_SPI_EXPECT_CLK_SPEED=500000'",
      "'-D ST7701_IO_3WIRE_SPI_SPI_MODE=SPI_MODE0'",
      "'-D ST7701_IO_3WIRE_SPI_LCD_CMD_BYTES=1'",
      "'-D ST7701_IO_3WIRE_SPI_LCD_PARAM_BYTES=1'",
      "'-D ST7701_IO_3WIRE_SPI_FLAGS_USE_DC_BIT=1'",
      "'-D ST7701_IO_3WIRE_SPI_FLAGS_DC_ZERO_ON_DATA=0'",
      "'-D ST7701_IO_3WIRE_SPI_FLAGS_LSB_FIRST=0'",
      "'-D ST7701_IO_3WIRE_SPI_FLAGS_CS_HIGH_ACTIVE=0'",
      "'-D ST7701_IO_3WIRE_SPI_FLAGS_DEL_KEEP_CS_INACTIVE=1'",
      "'-D ST7701_PANEL_CONFIG_CLK_SRC=LCD_CLK_SRC_PLL160M'", 
      "'-D ST7701_PANEL_CONFIG_TIMINGS_PCLK_HZ=(16*1000000)'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_H_RES=DISPLAY_WIDTH'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_V_RES=DISPLAY_HEIGHT'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_HSYNC_PULSE_WIDTH=8'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_HSYNC_BACK_PORCH=50'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_HSYNC_FRONT_PORCH=10'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_VSYNC_PULSE_WIDTH=8'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_VSYNC_BACK_PORCH=20'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_VSYNC_FRONT_PORCH=10'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_FLAGS_HSYNC_IDLE_LOW=false'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_FLAGS_VSYNC_IDLE_LOW=false'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_FLAGS_DE_IDLE_HIGH=false'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_ACTIVE_NEG=0'",
      "'-D ST7701_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_IDLE_HIGH=false'",
      "'-D ST7701_PANEL_CONFIG_DATA_WIDTH=16'",
      "'-D ST7701_PANEL_CONFIG_SRAM_TRANS_ALIGN=4'",
      "'-D ST7701_PANEL_CONFIG_PSRAM_TRANS_ALIGN=64'",
      "'-D ST7701_PANEL_CONFIG_HSYNC_GPIO_NUM=38'",
      "'-D ST7701_PANEL_CONFIG_VSYNC_GPIO_NUM=39'",
      "'-D ST7701_PANEL_CONFIG_DE_GPIO_NUM=40'",
      "'-D ST7701_PANEL_CONFIG_PCLK_GPIO_NUM=41'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_R0=46'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_R1=3'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_R2=8'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_R3=18'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_R4=17'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_G0=14'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_G1=13'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_G2=12'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_G3=11'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_G4=10'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_G5=9'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_B0=5'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_B1=45'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_B2=48'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_B3=47'",
      "'-D ST7701_PANEL_CONFIG_DATA_GPIO_B4=21'",
      "'-D ST7701_PANEL_CONFIG_DISP_GPIO_NUM=GPIO_NUM_NC'",
      "'-D ST7701_PANEL_CONFIG_FLAGS_DISP_ACTIVE_LOW=false'",
      "'-D ST7701_PANEL_CONFIG_FLAGS_RELAX_ON_IDLE=false'",
      "'-D ST7701_PANEL_CONFIG_FLAGS_FB_IN_PSRAM=true'",
      "'-D ST7701_DEV_CONFIG_RESET_GPIO_NUM=GPIO_NUM_NC'",
      "'-D ST7701_DEV_CONFIG_COLOR_SPACE=ESP_LCD_COLOR_SPACE_BGR'",
      "'-D ST7701_DEV_CONFIG_BITS_PER_PIXEL=16'",
      "'-D ST7701_DEV_CONFIG_FLAGS_RESET_ACTIVE_HIGH=false'",
      "'-D ST7701_DEV_CONFIG_VENDOR_CONFIG=NULL'",
      "'-D DISPLAY_SWAP_XY=false'",
      "'-D DISPLAY_MIRROR_X=false'",
      "'-D DISPLAY_MIRROR_Y=false'",
      "'-D BOARD_HAS_TOUCH'",
      "'-D CONFIG_ESP_LCD_TOUCH_MAX_POINTS=5'",
      "'-D CONFIG_ESP_LCD_TOUCH_MAX_BUTTONS=0'",
      "'-D TOUCH_GT911_I2C'",
      "'-D GT911_I2C_HOST=I2C_NUM_0'",
      "'-D GT911_I2C_CONFIG_SDA_IO_NUM=15'",
      "'-D GT911_I2C_CONFIG_SCL_IO_NUM=7'",
      "'-D GT911_I2C_CONFIG_SDA_PULLUP_EN=GPIO_PULLUP_ENABLE'",
      "'-D GT911_I2C_CONFIG_SCL_PULLUP_EN=GPIO_PULLUP_ENABLE'",
      "'-D GT911_I2C_CONFIG_MASTER_CLK_SPEED=400000'",
      "'-D GT911_I2C_CONFIG_CLK_FLAGS=I2C_SCLK_SRC_FLAG_FOR_NOMAL'",
      "'-D GT911_IO_I2C_CONFIG_DEV_ADDR=GT911_IO_I2C_CONFIG_DEV_ADDRESS_5D'",
      "'-D GT911_IO_I2C_CONFIG_CONTROL_PHASE_BYTES=1'",
      "'-D GT911_IO_I2C_CONFIG_DC_BIT_OFFSET=0'",
      "'-D GT911_IO_I2C_CONFIG_LCD_CMD_BITS=16'",
      "'-D GT911_IO_I2C_CONFIG_LCD_PARAM_BITS=0'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DC_LOW_ON_DATA=false'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DISABLE_CONTROL_PHASE=true'",
      "'-D GT911_TOUCH_CONFIG_X_MAX=DISPLAY_WIDTH'",
      "'-D GT911_TOUCH_CONFIG_Y_MAX=DISPLAY_HEIGHT'",
      "'-D GT911_TOUCH_CONFIG_RST_GPIO_NUM=GPIO_NUM_NC'",
      "'-D GT911_TOUCH_CONFIG_INT_GPIO_NUM=GPIO_NUM_NC'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_RESET=0'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_INTERRUPT=0'",
      "'-D TOUCH_SWAP_XY=false'",
      "'-D TOUCH_MIRROR_X=false'",
      "'-D TOUCH_MIRROR_Y=false'"

    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "debug": {
    "openocd_board": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "esp32-s3-touch-lcd-4",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4",
  "vendor": "Waveshare"
}

I get an output on the display but the colours are completely wrong.

What I'm getting...
[image]

If i change       "'-D ST7701_DEV_CONFIG_COLOR_SPACE=ESP_LCD_COLOR_SPACE_BGR'", to RGB i get the following but still wrong.
[image]

What it should look like...
[image]

I have triple checked my RGB GPIO Pin assignments and they are correct. Is there another setting somewhere im missing?

Thanks :)

rzeldent, 2025-02-24:
Look at the schema to see if the data is correct: So tweaking

"'-D ST7701_PANEL_CONFIG_DATA_GPIO_R0=46'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_R1=3'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_R2=8'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_R3=18'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_R4=17'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_G0=14'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_G1=13'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_G2=12'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_G3=11'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_G4=10'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_G5=9'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_B0=5'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_B1=45'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_B2=48'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_B3=47'",
"'-D ST7701_PANEL_CONFIG_DATA_GPIO_B4=21'",

gavzzorz, 2025-02-25:
I followed the schematic to the T and tripple checked it. also checked the schematic is correct from the pin assignments in the Waveshare demo project.

Also checked im setting the GPIO numbers to the correct Panel Config assignments in the boards file.
To me everything looks correct but im new to this ESP32 Display stuff.

[image]

[image]

Also from the example library this is the pin assignment that matches everything i have set in the Boards file.
    46 /* R0 */, 3 /* R1 */, 8 /* R2 */, 18 /* R3 */, 17 /* R4 */,
    14 /* G0 */, 13 /* G1 */, 12 /* G2 */, 11 /* G3 */, 10 /* G4 */, 9 /* G5 */,
    5 /* B0 */, 45 /* B1 */, 48 /* B2 */, 47 /* B3 */, 21 /* B4 */,

rzeldent, 2025-02-25:
Is it maybe an IPS panel? Then also define DISPLAY_IPS=1

gavzzorz, 2025-02-25:
It is an IPS Display so i will give that a go thankyou.
I based my esp32-s3touchlcd4.json on the existing esp32-s3touchlcd7.json that is similar in a lot of way's That 7" Display is also IPS and RGB but that doesnt define DISPLAY_IPS=1 in the boards file.

gavzzorz, 2025-02-25:
Ok tried defining DISPLAY_IPS=1 and it does make a difference but still not right.
I tried both RGB and BGR Colour Space, With RGB Red is displayed correctly but the other two colours are wrong.

[image]
[image]

now im lost and confused haha

gavzzorz, 2025-03-04:
> > Colour Space, With RGB Red is displayed correctly but the other two colours are wrong.
> 
> Do you know if there is any progress? Looking forward to fixing this issue with waveshare esp32 s3 touch LCD 4 inch with this lib.

Do you have the same display with the same issue then? not resolved for me no, iv tried all i can think of with no luck. need someone with more knowledge to pitch in.

gavzzorz, 2025-03-07:
Nice to know theres someone else on it with me. keep this thread updated if you have any progress.

I also had an issue with the rebooting error on my main project using this boards json and i think thats down to an issue with my main project as when i moved to a test blank project using the same boards file i then dont get the reboot.

betamoojw, 2025-03-12:
> Nice to know theres someone else on it with me. keep this thread updated if you have any progress.
>
> I also had an issue with the rebooting error on my main project using this boards json and i think [...]

I've created the [Waveshare_ESP32-S3-Touch-LCD-4Inch](https://github.com/betamoojw/Waveshare_ESP32-S3-Touch-LCD-4Inch) project as startup by using the ESP32_Display_Panel and ESP32_IO_Expander libraries instead of this esp32-smartdisplay lib, and it works fine as expected. Hope it's helpful for you.

rzeldent, 2025-03-15:
Hi Gavorzz,

I think there is still something not right with the wiring definition. If I take a look at the first two images, the G and B have kinda the same color. Is the schematic correct?

digiexchris, 2025-03-24:
This may not be the cause of your issue, as I haven't gotten that far on a similar display, but on a different display using LVGL with a different framework, I had weird colour issues due to pixel data being big endian on the display but SPI on most microcontrollers are sending 8 bits at a time little endian. Effectively, a colour 0xA3F4 was being recieved as 0xF4A3 by the display. setting LV_COLOR_16_SWAP in lv_conf.h fixed it for me.

rzeldent, 2025-03-31:
Hi Gavorzz / digiexChris,

This LV_COLOR_16_SWAP  does no longer exists in LVGL so if this is the culprit, it might be be resolved by swapping the order in the buffer in direct_io_lv_flush. Maybe also changing the mapping of the color bits might help..

digiexchris, 2025-04-04:
https://github.com/lvgl/lvgl/blob/5c5ae5d2bc9b3626bfed246a8533a1b1a3aeba20/src/core/lv_refr.c#L1320 it does exist for backwards compatibility (linked to master), but apparently there is another way to accomplish the same thing in newer versions.

gavzzorz, 2025-04-04:
To update, I have tried all the above with no success. I'm sure of the schematic being correct as the exact same pin assignment is used for their assignments in the Arduino examples and that displays correctly.

ASD123-dsh, 2025-04-19:
Hello. May I ask if this issue has been resolved now? I have also encountered this problem. When I was debugging a screen with the ST7701 IC, which has a 3SPI + 18RGB interface and a resolution of 480*480, the screen was successfully lit up, but the colors were completely incorrect.

closed 2026-10-09 by github-actions[bot]