rzeldent/esp32-smartdisplay #249: I2C External bus for the board esp32-s3touchlcd7

Sanjaypatel145, 2025-03-20:
demo works seamlessly with esp32-s3touchlcd7,

help needed to integrate with other i2c sensors

though board is configured as 
      "'-D TOUCH_GT911_I2C'",
      "'-D GT911_I2C_HOST=I2C_NUM_0'",
      "'-D GT911_I2C_CONFIG_SDA_IO_NUM=8'",
      "'-D GT911_I2C_CONFIG_SCL_IO_NUM=9'",
      "'-D GT911_I2C_CONFIG_SDA_PULLUP_EN=GPIO_PULLUP_ENABLE'",
      "'-D GT911_I2C_CONFIG_SCL_PULLUP_EN=GPIO_PULLUP_ENABLE'",
      "'-D GT911_I2C_CONFIG_MASTER_CLK_SPEED=400000'",
      "'-D GT911_I2C_CONFIG_CLK_FLAGS=I2C_SCLK_SRC_FLAG_FOR_NOMAL'",
      "'-D GT911_IO_I2C_CONFIG_DEV_ADDR=GT911_IO_I2C_CONFIG_DEV_ADDRESS_5D'",
      "'-D GT911_IO_I2C_CONFIG_CONTROL_PHASE_BYTES=1'",
      "'-D GT911_IO_I2C_CONFIG_DC_BIT_OFFSET=0'",
      "'-D GT911_IO_I2C_CONFIG_LCD_CMD_BITS=16'",
      "'-D GT911_IO_I2C_CONFIG_LCD_PARAM_BITS=0'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DC_LOW_ON_DATA=false'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DISABLE_CONTROL_PHASE=true'",

      "'-D GT911_TOUCH_CONFIG_INT_GPIO_NUM=GPIO_NUM_NC'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_RESET=0'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_INTERRUPT=0'",

rzeldent, 2025-03-21:
Hi Sanjay,

To use other sensors you'll need to create a new driver. This can be done but is not just changing some defines.
What are you wanting to integrate?

Sanjaypatel145, 2025-03-22:
It is NAU7802 adc and drivers are [SparkFun_Qwiic_Scale_NAU7802_Arduino_Library](https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library)

rzeldent, 2025-03-22:
You can just add these libraries independent of the definitions. The I2C bus can handle multiple devices no should not be a problem

closed 2026-10-09 by github-actions[bot]