rzeldent/esp32-smartdisplay #250: esp_lcd_rgb_panel_config_t has no member errors

hillsandales, 2025-03-24:
https://github.com/rzeldent/esp32-smartdisplay/blame/d6c8c7105b78c547d03bf8a7c5372ff27604748a/src/lvgl_panel_st7262_par.c#L105C59-L107

I'm getting build errors for struct `esp_lcd_rgb_panel_config_t`.

It looks like three members were added (expected) to the struct in this repo, specifically in `lvgl_panel_st7262_parc` that I'm using.

However, it doesn't look like the official `esp_lcd` support that defines the struct (`esp_lcd_panel_rgb.h`) defines these members.

https://github.com/espressif/esp-idf/blob/ee77c489e73993493e95bf765c0955cb87d59166/components/esp_lcd/rgb/include/esp_lcd_panel_rgb.h#L175

The members in question are:

1. `.on_frame_trans_done`
2. `.user_ctx`
3.  `.relax_on_idle` (under the `.flags` member)

The history shows these were added at least in part for LVGL v9. I'm building against LVGL v9.2.0 and v9.2.2, but neither works. I'm using esp-idf v5.3.2 due to other dependencies like Arduino not working on the latest v5.4 version.

Is this a mistake or am I doing something wrong? Removing these members allowed it to build.

vanbwodonk, 2025-03-26:
Hei @hillsandales, i have same problems. I fix it with downgrading espressif platform
```
platform = espressif32@6.9.0
```

rzeldent, 2025-03-31:
Hi, I retried it on the **develop** branch:

after a ```pio pkg update```:
Platform Manager: espressif32@6.9.0 has been removed!
Platform Manager: Installing platformio/espressif32 @ 6.10.0

Library Manager: Installing lvgl/lvgl @ ^9.2.2

Environment      Status    Duration
---------------  --------  ------------
esp32-8048S043C  SUCCESS   00:03:48.868

olivluca, 2025-04-22:
```
Library Manager: lvgl@9.2.2 has been removed!
Library Manager: esp32_smartdisplay@2.1.0+sha.a49bf44 has been removed!
Library Manager: Installing git+https://github.com/rzeldent/esp32-smartdisplay.git#develop
git version 2.39.5
S'està clonant a «/home/luca/.platformio/.cache/tmp/pkg-installing-5b9148cl»...

[...]

Library Manager: esp32_smartdisplay@2.1.0+sha.a305459 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing lvgl/lvgl @ ^9.2.2
Unpacking  [####################################]  100%
Library Manager: lvgl@9.2.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-2432S028R.html
PLATFORM: Espressif 32 (51.3.6+sha.ec17486) > esp32-2432S028R
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.7 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.632e0c2a9f 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 2.1.0+sha.a305459
|-- ESP32MQTTClient @ 1.0.0
|-- ArduinoOTA @ 3.0.7
|-- WiFi @ 3.0.7
|-- lvgl @ 9.2.2
Building in release mode
Compiling .pio/build/trumadisplay/src/main.cpp.o

[...]

Compiling .pio/build/trumadisplay/lib057/esp32_smartdisplay/esp_panel_st7796.c.o
.pio/libdeps/trumadisplay/esp32_smartdisplay/src/esp_panel_ili9341.c: In function 'esp_lcd_new_panel_ili9341':
.pio/libdeps/trumadisplay/esp32_smartdisplay/src/esp_panel_ili9341.c:374:14: error: 'esp_lcd_panel_t' has no member named 'disp_off'; did you mean 'disp_on_off'?
  374 |     ph->base.disp_off = ili9341_disp_off;
      |              ^~~~~~~~
      |              disp_on_off
Compiling .pio/build/trumadisplay/lib057/esp32_smartdisplay/esp_touch_cst816s.c.o
*** [.pio/build/trumadisplay/lib057/esp32_smartdisplay/esp_panel_ili9341.c.o] Error 1
======================================= [FAILED] Took 104.26 seconds =======================================

```

olivluca, 2025-04-22:
(probably this is a different issue, but still related to the updated platform, three cheers for backwards compatibility 😉 )

olivluca, 2025-04-23:
Strange, disp_off was removed 3 years ago
https://github.com/espressif/esp-idf/commit/de433105a28f87517f726fb11d26ad0a671ce863#diff-24cb520ef03f7d86bf534128937c63564b799ad8d2da55d6157cf4ceda3c7fe0

olivluca, 2025-04-26:
It turns out I was using a git version (which I installed and forgot) of the espressif32 platform instead of 6.9.0/6.10.0 🤦