rzeldent/esp32-smartdisplay #263: Access to expander

zejulio, 2025-07-04:
Hello,

I am using the  Waveshare ESP32-S3-Touch-LCD-7 with your library.
How can I access the expander?
I need to set the USB/UART to CAN...

Sometimes I also get blank screen, I need to put the example, and then back my software for it to work. I believe that is something related to the expander setup.

Thanks for any tip,

JS

zejulio, 2025-07-07:
@rzeldent
Can you help me on this?

rzeldent, 2025-07-07:
See: https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-7/ESP32-S3-Touch-LCD-7-Sch.pdf

The CAN ports are available on IO19 (CANRX) and IO20 (CANTX)

[image]

zejulio, 2025-07-07:
Yes, but they are multiplexed by U9 PIN 10 (USB_SEL) and that is on U10 CH422G IO5 (EXIO5).
So I need to set IO5 to 1 on U10 before I can use CAN_TX/RX, else it will be connected to USB pins.
I've managed to do that with ESP_IOExpander_Library, but when using this library your's stop working.
So I apply power, use this library to set IO5 HIGH, then comment and use your lib...

Also when using your lib I get this:
[   244][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (66) gpio: gpio_set_level(227): GPIO output gpio_num error
E (66) ledc: ledc_channel_config(608): gpio_num argument is invalid
[   275][W][esp_touch_gt911.c:87] gt911_reset(): No RST pin defined

The GT911 is also on U10 CH422G IO1 (CTP_RST)
And the LCD displays is also on U10 CH422G IO3 (LCD_RST)
And... also the LCD power is on U10 CH422G IO6 (LCD_VDD_EN)
In your list this signals from U10 show at end EXIOx, and I thought your library had this in mind and set the U10 CH422 properly.
But I found out that sometimes the screen goes black when powering ON, and found out that IO6 is LOW, so the LCD power is OFF, then the LCD does not work...
Can you please add something that we can use to control the CH422G?
In the original waveshare demo's they call it "expander".
I belive when I initialize the expander, it mess up with "wire"/I2C, and then the touch panel does not initialize and that's the issue when using both libraries.

Can you please help?

Thanks!

zejulio, 2025-07-07:
[image]

rzeldent, 2025-07-07:
Why use an expander library? Just set IO05 to High so you'll have always access to the CAN.

zejulio, 2025-07-07:
@rzeldent iO5 is on CH422 I2C port expander, not on ESP32!
I need to set CH422 IO5 HiGH thru I2C... thats why i need to use I2C...

zejulio, 2025-07-07:
Correct. Since i use the usb2serial to flash and debug, i dont need the esp32 usb, so i keep it always high.
Ive managed to work like this, as explained before.
The issue is that if I use I2C before your library, your library dont work and panic when using i2c for touch...
So I need to use the same driver your library use to connect to i2c bus. Or other kind of alternative to share the use of i2c bus.
The i2c address is different so no problem.
The problem i have is to share on the code the i2c bus on both libraries...

Also as told, in this board, sometimes your library fail to work cause does not initiate properly the expander and it could start with LCD-VCC-EN LOw and this way the LCD has no power...
You need to initialize the expander with most of the outputs HIGH, like the VCC_EN or the reset for LCD and Touch.
All pins tagged as EXIO are connected to expander and you need i2c to set them...

I dont know if this is the only board using this...
Whay is your idea?

closed 2026-10-09 by github-actions[bot]