rzeldent/esp32-smartdisplay #265: Latest develop branch doesn't compile on my machine

bobskigit, 2025-07-06:
Using the latest development branch I get compile errors

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-3248S035C.html
PLATFORM: Espressif 32 (54.3.20) > esp32-3248S035C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.0 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a 
 - tool-esptoolpy @ 4.8.9 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 9.3.0
|-- test @ 0.0.0+20250706084122

src/esp_panel_st7796.c: In function 'esp_lcd_new_panel_st7796':
src/esp_panel_st7796.c:355:14: error: 'esp_lcd_panel_t' has no member named 'disp_off'; did you mean 'disp_on_off'?
  355 |     ph->base.disp_off = st7796_disp_off;
      |              ^~~~~~~~
      |              disp_on_off
Compiling .pio\build\esp32-3248S035C\src\lvgl_touch_gt911_i2c.c.o
*** [.pio\build\esp32-3248S035C\src\esp_panel_st7796.c.o] Error 1
src/esp32_smartdisplay_dma.c: In function 'smartdisplay_dma_transfer_chunk':
src/esp32_smartdisplay_dma.c:361:29: error: implicit declaration of function '_min' [-Wimplicit-function-declaration]
  361 |         size_t chunk_rows = _min(remaining / bytes_per_row, g_dma_manager->dma_buffer_size / bytes_per_row);
      |                             ^~~~

bobskigit, 2025-07-06:
I'm on a Windows machine and  this is the dependency map

platformio.exe pkg list --environment esp32-1732S019C 

Resolving esp32-1732S019C dependencies...
Platform espressif32 @ 54.3.20 (required: espressif32)
├── framework-arduinoespressif32 @ 3.2.0 (required: https://github.com/espressif/arduino-esp32/releases/download/3.2.0/esp32-3.2.0.zip)
├── framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a (required: https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.4/esp32-arduino-libs-idf-release_v5.4-2f7dcd86-v1.zip)
├── tool-esptoolpy @ 4.8.9 (required: https://github.com/pioarduino/esptool/releases/download/v4.8.9/esptool.zip)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 3.2.0 (required: tasmota/tool-mklittlefs @ ^3.2.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 (required: platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403)
├── tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 (required: platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403)
├── toolchain-riscv32-esp @ 14.2.0+20241119 (required: platformio/toolchain-riscv32-esp @ 14.2.0+20241119)
└── toolchain-xtensa-esp-elf @ 14.2.0+20241119 (required: platformio/toolchain-xtensa-esp-elf @ 14.2.0+20241119)

Libraries
├── lvgl @ 9.3.0 (required: lvgl/lvgl @ ^9.2.2)
└── test @ 0.0.0+20250706183826 (required: file://C:\Users\bob.lamb\Documents\projects\esp32-smartdisplay\test)

bobskigit, 2025-07-06:
Moving to the develop branch of the boards submodule works for esp32-1732S019C, but esp32-3248S035C or R  still fail with 

Compiling .pio\build\esp32-3248S035C\src\esp_panel_st7796.c.o
src/esp_panel_st7796.c: In function 'esp_lcd_new_panel_st7796':
src/esp_panel_st7796.c:355:14: error: 'esp_lcd_panel_t' has no member named 'disp_off'; did you mean 'disp_on_off'?
  355 |     ph->base.disp_off = st7796_disp_off;
      |              ^~~~~~~~
      |              disp_on_off

bobskigit, 2025-07-06:
Thanks Rene, that fixes it.

Should we update the **platformio.ini** dependency to platform =  espressif32 @ 6.11.0

Bob

zejulio, 2025-07-07:
I can only compile it with version 6.9, if above got seral issues.