rzeldent/esp32-smartdisplay #267: ESP32-s3touchlcd4p3 - LVGL Touch Init boot loop

Cgriz365, 2025-07-19:
During most boots I get the following error and information in my monitor along with a continuous boot loop: (Please note that the first 5 lines are not part of the error and appear every time the code runs).
```

[   381][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   412][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (38) gpio: gpio_set_level(227): GPIO output gpio_num error
E (38) ledc: ledc_channel_config(608): gpio_num argument is invalid
[   443][W][esp_touch_gt911.c:87] gt911_reset(): No RST pin defined
E (70) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
[   450][E][esp_touch_gt911.c:143] gt911_read_info(): Unable to read GT911_PRODUCT_ID_REG
[   465][E][esp_touch_gt911.c:432] esp_lcd_touch_new_i2c_gt911(): GT911 read info failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4037d4a0
file: ".pio/libdeps/esp32-s3touchlcd4p3/esp32_smartdisplay/src/lvgl_touch_gt911_i2c.c" line 78
func: lvgl_touch_init
expression: esp_lcd_touch_new_i2c_gt911(io_handle, &touch_config, &touch_handle)

Backtrace:0x40377ff6:0x3fcb9920 0x4037d4ad:0x3fcb9940 0x40383ae5:0x3fcb9960 0x4037d4a3:0x3fcb99e0 0x4204331b:0x3fcb9a00 0x42042f66:0x3fcb9a90 0x42001a23:0x3fcb9ab0       

  #0  0x40377ff6 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  #1  0x4037d4ad in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
  #2  0x40383ae5 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
  #3  0x4037d4a3 in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:43
  #4  0x4204331b in lvgl_touch_init at .pio/libdeps/esp32-s3touchlcd4p3/esp32_smartdisplay/src/lvgl_touch_gt911_i2c.c:48 (discriminator 1)
  #5  0x42042f66 in smartdisplay_init at .pio/libdeps/esp32-s3touchlcd4p3/esp32_smartdisplay/src/esp32_smartdisplay.c:210
  #6  0x42001a23 in lvgl_task(void*) at src/main.cpp:37
```
I am running LVGL through a task on core 1 of my Waveshare 4.3B. I am running another task handling wifi and other peripherals on core 0. Please note that the wifi task is initiated after the LVGL and display task. You can find my task code here: 
```
void lvgl_task(void *parameter)
{
  smartdisplay_init();

  __attribute__((unused)) auto disp = lv_disp_get_default();
  lv_disp_set_rotation(disp, LV_DISP_ROTATION_0);

  init_lvgl_littlefs_driver();
  ui_init();

  auto lv_last_tick = millis();

  for (;;)
  {
    auto const now = millis();

    lv_tick_inc(now - lv_last_tick);
    lv_last_tick = now;

    lv_timer_handler();
    vTaskDelay(5 / portTICK_PERIOD_MS);
  }
} 

xTaskCreatePinnedToCore(
      lvgl_task,
      "LVGL_Task",
      32000,
      NULL,
      15,
      &lvgl_task_handle,
      1);

```

I have my task running at a higher priority in hopes that there would not be interference but the issue persists at any priority. Here is my platform.ini file: 
```
[platformio]
default_envs = esp32-s3touchlcd4p3

[env]
platform = espressif32@6.11.0
framework = arduino

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_filters = esp32_exception_decoder

build_flags =
    -Ofast
    -Wall
    '-D BOARD_NAME="${this.board}"'
    '-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_INFO'
    #'-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG'
    #'-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE'
    '-D LV_CONF_PATH="${platformio.include_dir}/lv_conf.h"'
    '-DCONFIG_ESP_TASK_WDT_TIMEOUT_S=60'
    
board_build.partitions = default_8MB_FFAT.csv
board_build.filesystem = littlefs
    
lib_deps =
    https://github.com/rzeldent/esp32-smartdisplay/archive/refs/heads/main.zip
    mobizt/FirebaseClient @ ^2.1.5
    bblanchon/ArduinoJson @ ^7.4.2

[env:esp32-s3touchlcd4p3]
board = esp32-s3touchlcd4p3
```

Cgriz365, 2025-07-19:
Interesting update: 

After letting the board sit for around 15 minutes after code upload, this issue stops and the display is able to properly initiate along with the touch. I am unsure why this is and would like to be able to run my code directly after upload.

Cgriz365, 2025-07-20:
After further investigation into board definitions, it appears that both the touch_rst and touch_int are undefined. I cross referenced with this library here: (https://github.com/nishad2m8/WS-4.3/tree/main) and found that there is a touch_int definition. I've updated my board definition by changing this line:
`"'-D GT911_TOUCH_CONFIG_INT=GPIO_NUM_4'",`
This has made no change to the boot loop error unfortunately.

rzeldent, 2025-07-20:
I can see some errors regarding undefined pins. I do not own the board and obtained this configuration from others. Looks like the pins are not correctly defined. This should be fixed first.

closed 2026-10-09 by github-actions[bot]