rzeldent/esp32-smartdisplay #268: lvgl 9.2.2 no longer exists

gopro2027, 2025-07-26:
Title... lvgl 9.2.2 does not appear to exist anymore. When you install this it installs 9.3.0. This breaks my code. See log here:

[image]

You can see even though it attempts to install 9.2.2, at the end it clearly says 9.3.0 has been installed.

Furthermore, when you build your project it also shows 9.3.0

[image]

Any ideas on how to fix this?

gopro2027, 2025-07-26:
Okay so we solved it, this is the solution. You can't setup the conf_path like that old way anymore. Now you need to do:
```
    -I include  ; Add the include/ folder to the search path
    -DLV_CONF_PATH=\"lv_conf.h\"  ;Define LV_CONF_PATH as a quoted filename
```

Oddly, it's very specific it seems. At first I tried without the backslashes around the lv_conf.h and it didn't work. After doing a file compare with my friends working platformio.ini I saw he had backslashes so I added it and I fixed it. Why? no clue. But it compiles on 9.3.0 now. 
Ironically, I'll be sticking to 9.2.2 for the time being anyways because it uses less memory and I'm over 100% with 9.3.0

Nimajine, 2025-08-15:
Thank you so much! You've put your finger on a problem I've been struggling to solve for a long time! Congratulations on your insight.

closed 2026-10-09 by github-actions[bot]