rzeldent/esp32-smartdisplay #269: Cannot find GPIO_BCKL and sw_rotate wlike compiling for ESP32-2032S022C

chitkosarvesh, 2025-07-27:
```
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/esp_panel_gc9a01.c.o
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/esp_panel_ili9341.c.o
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/esp_panel_st7701.c.o
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/esp_panel_st7796.c.o
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/esp_touch_cst816s.c.o
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/esp_touch_gt911.c.o
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/esp_touch_xpt2046.c.o
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/lvgl_panel_gc9a01_spi.c.o
.pio/libdeps/esp32-2432S022C/esp32_smartdisplay/src/esp32_smartdisplay.c: In function 'smartdisplay_init':
.pio/libdeps/esp32-2432S022C/esp32_smartdisplay/src/esp32_smartdisplay.c:186:11: error: 'GPIO_BCKL' undeclared (first use in this function); did you mean 'GPIO_BT_SEL'?
   pinMode(GPIO_BCKL, OUTPUT);
           ^~~~~~~~~
           GPIO_BT_SEL
.pio/libdeps/esp32-2432S022C/esp32_smartdisplay/src/esp32_smartdisplay.c:186:11: note: each undeclared identifier is reported only once for each function it appears in
.pio/libdeps/esp32-2432S022C/esp32_smartdisplay/src/esp32_smartdisplay.c:197:15: error: 'lv_display_t' {aka 'struct _lv_display_t'} has no member named 'sw_rotate'
   if (!display->sw_rotate)
               ^~
Compiling .pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/lvgl_panel_ili9341_spi.c.o
*** [.pio/build/esp32-2432S022C/lib2be/esp32_smartdisplay/esp32_smartdisplay.c.o] Error 1
Indexing .pio/build/esp32-2432S022C/lib570/liblvgl.a
======================================================================================================================================================= [FAILED] Took 25.93 seconds =======================================================================================================================================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

chitkosarvesh, 2025-07-27:
I used lv_conf from version 9.1.0 and that seems to work okay.
Also, configuration should be divided into [env] and [env:esp32-2032S022C] separately as described in the example repo.

closed 2025-07-27 by chitkosarvesh

rzeldent, 2025-07-27:
Have you updated the boards?

```sh
git submodule update
```

rik43, 2025-08-15:
I had the same problem (with ESP32-3248S035C), after studying the library code I realized that it was an old version.

I replaced
lib_deps = rzeldent/esp32_smartdisplay
with the Github repository:
lib_deps = https://github.com/rzeldent/esp32-smartdisplay.git

and everything worked!